toyoshim/iona-us #34: Power Stone 1

Shentokk, 2022-06-27:
When using a PS4 controller, the game does not recognize the Iona as being connected. It seems to work fine with Xbox One controllers.

toyoshim, 2022-06-29:
Hum.... curious.
USB and JVS are logically isolated, and would not affect each other....

Shentokk, 2022-07-04:
[image]
It seems this happens on Jambo Safari too after a few minutes of play. Only with the PS4 controller used. Other controllers aren't having this issue.

toyoshim, 2023-05-05:
Let me merge this topic to #42 so that I can ensure if such stability issue still exist in v2 firmware

closed 2023-05-05 by toyoshim