toyoshim/iona-us #42: I/O errors on some naomi titles with 2 controllers

toyoshim, 2022-10-05:
Reported from [@Xeno_Dragon](https://twitter.com/Xeno_Dragon)

Before 1.45, there was an input delay issue on the same situation.
After 1.46, now it causes an I/O error after playing games for a while.

Conditions:
- PS4 controllers x2
- naomi or naomi2
- runs one of following titles (time indicates how long does it run without any error)
  * MvC2 (1m14s)
  * virtua tennis2 (15s)
  * cannonspike (13s)
  * gigawing 2 (1m7s)
  * virtua tennis (19s)
  * sega tetris (1m8s)

logan-mccloud, 2023-02-16:
Hello.

I have experienced something similar, in my case with the same firmware version, the solution was to switch to a PC controllers instead of a PS3/PS4 one.

If a use a PS3/PS4 controller, some games crash randomly or fail to launch, but with an standard PC wired controller it doesn´t happen.

Best regards.

toyoshim, 2023-05-05:
Thank you the report.
I suspect some PS controllers sent longer HID report and it may cause JVS side packet lost due to the device getting busy to handle it.
As new v2 firmware can achieve more standardized process to handle several controllers and assignments, it may mitigate the issue, or if we can see the same problem here, I will spend some time to invent a new mechanisms to avoid this kind of problems, e.g. using an interrupt to handle JVS request, or so.
So, if you can try v2 firmware, that's really appreciated.

Shentokk, 2023-05-05:
The issue still persists with a PS4 controller when trying the Initial D Export series on v2.00. On boot it will display an IO BD error that doesn't occur when using an Xbox One or Nintendo Switch controller.

toyoshim, 2023-05-28:
I did bunch of optimization this weekend, and now the JVS response performance is drastically improved.
So, I expect this improvement contribute to resolve this kind of I/O error issues.
Please try v2.10 to see if the new code resolves the issue.

Shentokk, 2023-06-06:
Power Stone now works with the v2.10 update with a PS4 controller. Thank you for working on the problem.