toyoshim/iona-us #44: Firmware 1.47 doesn´t work properly on SEGA Ringedge 2

logan-mccloud, 2023-02-16:
Hello,

I have recently upgrade my firmware from 1.40 to 1.47 and I have noticed that the device it doesn´t work with a SEGA Ringedge 2 motherboard anymore.

The system is unable to initialize (white screen) under this particular firmware version (any game).

Comming back to 1.40 solved the problem.

Best regards.

toyoshim, 2023-02-19:
Thank you for the report.
If you can try other versions to identify which version introduced the specific problem, that will help.
Also, just in case, can you try both modes, signal adjustment ON and OFF?

VampireSavior, 2023-04-19:
> Hello,
>
> I have recently upgrade my firmware from 1.40 to 1.47 and I have noticed that the device it doesn´t work with a SEGA Ringedge 2 motherboard anymore.
>
> The system is unable to initialize [...]

May I ask, what settings are you using to get 1.40 working with the RE2? I can't seem to get it working no matter what.

toyoshim, 2023-05-05:
Now I'm focusing on v2 firmware to support more various kind of boards and titles flexibly.
As I'm just published the first experimental v2 firmware, can you try it?
I think the root cause would be USB transactions are too heavy and some packet lost happens on JVS, or recent updates add more functions support in JVS. So, JVS transaction gets heavy and motherboard cannot handle everything in a permitted time slice for 1 frame.

As v2 firmware can allow you to have minimum function supports easily, it may help to investigate the problem.

toyoshim, 2023-05-28:
After a large refactoring, JVS response performance is drastically improved in v2.10.
I guess this will solve most stability issues.

toyoshim, 2023-06-28:
let's close for now.
feel free to reopen or file another bug if it isn't working yet.

closed 2023-06-28 by toyoshim