toyoshim/iona-us #47: Force Feedback for Ridge Racer V

lorizzuoso, 2023-04-01:
Hello,I would like to know if it's possible to add a force feedback feature in order to play RRV and other games that require it.

Originally RRV required Drive Specific, Requires drive motor "V257 STR PCB" to function so if someone is open to implement this then I would gladly appreciate it.

toyoshim, 2023-04-01:
tweeter thread to gather information is here; https://twitter.com/toyoshim/status/1642183859986128896
but Eric said it required yet another interface to drive it (== STR PCB?)
so, I think we need to develop another STR PCB compatible hardware, or software patch to escape or convert it.

lorizzuoso, 2023-04-01:
I think that a software patch would be more convenient,I do know that the game reads the data from the dongle as all the code is present inside the "START" file in the dongle itself.