toyoshim/iona-us #54: Service button doesn't work in FW 2.0?

mathewbeall, 2023-05-13:
I am testing on a Namco System N2 (WMMT3)

Firmware 1.47 - service and test work fine (moving around the menu).

Firmware 2.0 - service doesn't work.

here is the cut and paste from the support page

"HIDI2C Device"
 VID: 0x056a
 PID: 0x494a
[Top Collections]
 #: 4
 [Collection 0]
  children#: 0
  feature#: 7
  input#: 1
  output#: 0
  input0: 0x85, 0x13, 0x95, 0x12, 0x75, 0x08, 0x06, 0x11, 0xff, 0x09, 0x00, 0x81, 0x02
 [Collection 1]
  children#: 0
  feature#: 1
  input#: 0
  output#: 0
 [Collection 2]
  children#: 0
  feature#: 5
  input#: 3
  output#: 0
  input0: 0x85, 0x0b, 0x95, 0x01, 0x75, 0x01, 0x06, 0x11, 0xff, 0x09, 0x42, 0x81, 0x02, 0x95, 0x01, 0x75, 0x01, 0x06, 0x11, 0xff, 0x09, 0x44, 0x81, 0x02, 0x95, 0x01, 0x75, 0x01, 0x06, 0x11, 0xff, 0x09, 0x45, 0x81, 0x02, 0x95, 0x01, 0x75, 0x01, 0x06, 0x11, 0xff, 0x09, 0x3c, 0x81, 0x02, 0x95, 0x01, 0x75, 0x01, 0x06, 0x11, 0xff, 0x09, 0x5a, 0x81, 0x02, 0x95, 0x01, 0x75, 0x01, 0x06, 0x11, 0xff, 0x09, 0x32, 0x81, 0x02, 0x95, 0x01, 0x75, 0x02, 0x81, 0x01, 0x95, 0x01, 0x75, 0x10, 0x05, 0x01, 0x09, 0x30, 0x81, 0x02, 0x95, 0x01, 0x75, 0x10, 0x05, 0x01, 0x09, 0x31, 0x81, 0x02, 0x95, 0x01, 0x75, 0x10, 0x05, 0x0d, 0x09, 0x30, 0x81, 0x02, 0x95, 0x01, 0x75, 0x10, 0x06, 0x00, 0xff, 0x09, 0x5b, 0x81, 0x02, 0x95, 0x01, 0x75, 0x20, 0x05, 0x0d, 0x09, 0x5b, 0x81, 0x02, 0x95, 0x01, 0x75, 0x08, 0x06, 0x00, 0xff, 0x09, 0x00, 0x81, 0x02, 0x95, 0x01, 0x75, 0x08, 0x05, 0x0d, 0x09, 0x3b, 0x81, 0x02, 0x95, 0x01, 0x75, 0x08, 0x05, 0x0d, 0x09, 0x3d, 0x81, 0x02, 0x95, 0x01, 0x75, 0x08, 0x05, 0x0d, 0x09, 0x3e, 0x81, 0x02, 0x95, 0x01, 0x75, 0x20, 0x81, 0x01
  input1: 0x85, 0x10, 0x95, 0x14, 0x75, 0x08, 0x06, 0x00, 0xff, 0x09, 0x00, 0x81, 0x02, 0x95, 0x01, 0x75, 0x08, 0x81, 0x01
  input2: 0x85, 0x1f, 0x95, 0x15, 0x75, 0x08, 0x06, 0x00, 0xff, 0x09, 0x00, 0x81, 0x02
 [Collection 3]
  children#: 0
  feature#: 1
  input#: 0
  output#: 0
END

toyoshim, 2023-05-19:
Thank you for your report.
Let me check this weekend!

mathewbeall, 2023-05-19:
Thank you!

Matt

toyoshim, 2023-05-19:
OK, I confirmed the problem and had a local fix now.
The problem happens because the new firmware updated the service button state only when it received controller's state from a USB gamepad. So, if there is no gamepad, the service button stops working.

I will push the next version 2.01 in a day, and it will include the fix!

closed 2023-05-19 by toyoshim

mathewbeall, 2023-05-19:
Thank you!

Matt

mathewbeall, 2023-05-20:
Works perfect - thank you!

Matt