toyoshim/iona-us #56: Officual support for Time Crisis 2

toyoshim, 2023-05-20:
Related discussion; https://twitter.com/toyoshim/status/1495424245694070784?s=20

Official I/O: namco ltd.;TSS-I/O;Ver2.02;JPN,GUN-EXTENSION
 - Switch 12bits
 - Coin 1 slot
 - Screen 16-bit x 1ch
 - Drive Out: 3 slot
 
|JVS|Time Crisis2|
|-|-|
|TEST|TEST|
|SERVICE|SERVICE|
|UP|MENU UP|
|DOWN|MENU DOWN|
|B1|Enter|
|B2|Trigger|
|B3|Peral|

toyoshim, 2023-06-13:
Almost working in the latest 2.12 firmware.
Adjustment seems a little flaky for Y axis.

pizzzza19, 2024-10-07:
こんにちは、自分の設定が間違っているかもしれませんがTSS-I/Oで16bitの設定を作りましたがFW 2.22ではネゴシエーションに失敗しサブボードエラーですすめませんでした。もし解決済でしたら各種設定ご教授いただければ幸いです。

toyoshim, 2024-10-07:
こちらにも書いておきます。

JVS ID: namco ltd;JYU-PCB
画面座標入力数: 1
画面座標入力幅: 16

割り当てはマップはこちらをご確認ください。
https://github.com/toyoshim/iona-us/wiki/Gun-Shooting

タイムクライシスは1〜3まで調査していた時期があったのですが、デバッグに付き合ってくれていた方の環境で、Guncon3照準がうまくキャリブレーションできない問題が解決できず、presetの公開には至っていない現状です。

pizzzza19, 2024-10-08:
ありがとうございます。Xboxコントローラーでは座標が読み取れず（白くフラッシュするだけ）基板がリセットを繰り返すのでガンコンでないと動かないかもしれません。もう少しJVS信号補正等設定をいじってみます。

pizzzza19, 2024-10-08:
特に変化はありませんでした。リブートを繰り返しているのは自分の基板が壊れかけているのが原因かもしれませんので、頂いた設定で通常動くのであれば一旦クローズで構いません（論点がぶれてしまうため）。検証・ご報告できず恐縮ですがご丁寧にご教授感謝いたします。ありがとうございました。

toyoshim, 2024-10-08:
なるほど、リブートの件を失念してました。
IONA-USの基板はv3系でしょうか、それ以前でしょうか？
v3なら電気特性的な相性はないと思うのですが、v1/v2でしたら電気特性で通信が不安定になりリブートに繋がることもあるかと思います。その場合、"JVSデータ信号レベル補正"をONとOFFで試して安定する方で使って下さい。
ケーブルやIONAの電源を変えて改善する事もあります。

あと、以前タイムクライシスのテストをしていた頃は、v2.02だったので、そのバージョンを試してもらうのも手かもしれません。v2.10の時に高速化をしてタイミングとか少し変わっているので、それが原因で相性が悪くなってる可能性もゼロではありません。

pizzzza19, 2024-10-08:
IONAはv3系でFWは2.22に自分でアップデートいたしました。
設定は：
_JVS ID: namco ltd;JYU-PCB
P1 アナログ 1(LX) -> 画面座標入力, 0
P1 アナログ 2(LY) -> 画面座標入力, 1
画面座標入力数: 1
画面座標入力幅: 16_
でJVSデータ信号レベル補正はどちらも差は無かったです。
本来は純正のv185サブボードを用意して正常動作するか切り分けが前提なのですが、あとはv2.02にFW下げて試してみます。それでだめであれば自分の基板がおかしいと方向づけできそうです。
NAOMI基板では問題なかったのでナムコ基板が信号にシビアなのか不安定な何か気配があります。またわかり次第追記いたします。詳細ありがとうございます。

pizzzza19, 2024-10-08:
v2.02 / v2.12 / v1.48 / v2.22でそれぞれ試してみましたが、基本的にバージョンが上がるほど認識しやすくなり自分の環境ではv2.12未満では全く動きませんでした。なんとなくファームウェアの相性な気もしますが手持ちの基板ではここまでが限界のため将来の解決を気長に待ちます。ありがとうございました

pizzzza19, 2025-04-18:
System 23 is JVS machine but USB port supported. You may prepare custom USB-A to front CON cable at yourself. As far as i know there is no such thing on the market. On my case it won't work. Not sure my system is broken or custom cable defective. System keeps rebooting.

toyoshim, 2025-04-19:
Yeah, JVS signals, SENSE, DATA-, DATA+, and GND are not exposed from a standard JVS (== USB) port, but just exposed from their own I/O connector directly. You can check their manuals, and IIRC, it contains diagrams that tell us which connectors provide these signals.
So, what you need is to wire them to a USB female connector, and connect it to IONA by a USB cable.