toyoshim/iona-us #60: [System 357] Keyboard Support?

anxious132, 2023-07-19:
Hey, as i cant find space to write it, i cant DM on twitter for some reason.

I just bought IONA, i would want to play Tekken 6 with keyboard is it possible with Mahjong mode?

toyoshim, 2023-08-19:
I didn't imagine that people may want to use keyboard for games other than Mahjong.
So, it didn't work as even if the IONA recognizes the keyboard, JVS side protocol is quite different.

e.g., 
https://github.com/toyoshim/iona/wiki/SW-Standard
https://github.com/toyoshim/iona/wiki/SW-SuchiePai (naomi's mahjong standard protocol that IONA support today)

toyoshim, 2023-08-19:
Let me mark this as a feature request.
It would not be difficult from the technical perspective, but I don't have a good idea to add a functionality to customize the keyboard map.

anxious132, 2023-08-19:
it would be cool to map keyboard as PS4 pad, iam using really common setup for Tekken games
WSAD for movement
Numpads for Attacks
 
 can i give my Map settings i use to be implemented?

anxious132, 2023-08-19:
i use bindings like this for NA-JV 
W - UP
A - LEFT
S - DOWN
D - RIGHT
-----------
Numpad 8 - 4
Numpad 5 - 2
Numpad 4 - 1
Numpad 6 - 3
Numpad 7 - Extra Right button