toyoshim/iona-us #62: Triforce issues.

PlatynumX, 2023-09-07:
My mp07 syncs with a triforce up using 1.47 firmware but not on any version of 2.xx. If I use the settings page to change Id to "namco najv",adjust controller button assignment and turn on data signal boost the device stops blinking but doesn't recognize any input from a controller. If I downgrade firmware to 1.47 "it just works". Help?

toyoshim, 2023-10-23:
Oh, can you try 2.11 or later?
There was a bug around bus reset handling, and it causes a compatibility issue with exA-Arcadia. So, it would not be surprising that it also affected triforce.
Also, each board implements their own board checks. So, can you try 2.11+ with the same ID and the feature set with the one you tried with 1.47?

PlatynumX, 2023-10-23:
2.11+ the device never syncs and seems to powers off (led turns off) if you go into the test menu of the triforce.

PlatynumX, 2023-10-23:
I had just flashed 2.17 and plugged in a Logitech g920. I never touched the settings. It shut off with the wheel and an Xbox elite series controller .I'm at work at the moment. I'll shoot a video later of it shutting off and tag you on Twitter. My handle is @ogplatynumx.

toyoshim, 2023-10-23:
Do you correctly setup the layout setting?
It needs to flash a proper setting from the layout setting page. There is no built-in default setup.
You need to craft a setting carefully for each game and controller.