toyoshim/iona-us #65: [System256]ファームウェア1.24以上だと動作しない。

Sonoda-a, 2023-11-22:
※XでDMをしたかったのですができなかったためissueから失礼します。

いつも大変お世話になっております。
toyoshim様の開発されたMP07-IONA-USは画期的であり、
我々の仲間内でも購入するように薦めております。

System256を用いて対戦ゲーム（機動戦士ガンダムseed destiny 連合vs.z.a.f.t.ii）をしているのですが、
おかげ様で競技人口が増え続けており、心より感謝致します。
また、Ver.3.00基板のリリースおめでとうございます。

そこで1つご相談なんですが、System256で接続する際にファームウェア1.24以上だと認識されない問題があり、
仲間内ではファームウェア1.20で使用するように声かけをさせて頂いております。
しかし、Ver.3.00基板からはファームウェア1.48または2.11以降しか動作しないとのことで
このままだと新規参入ができなくなってしまう状況です。

大変お手数をおかけしますが、最新のファームウェアでも動作できるように調査していただけませんか？
使用しているコントローラはHORIの「REAL ARCADE PRO V3-SA」です。

動作するようになれば仲間内にも連絡して購入するようにお勧めしますので
是非よろしくお願いします！

toyoshim, 2023-11-22:
ご連絡ありがとうございます。
最新版で壊れたままだった事、把握できていませんでした。
優先度上げて調査してみます。
なんとなく、その辺りにユーザーが多そうな事は把握していて、ゲームの方は入手してありますw

Sonoda-a, 2023-11-22:
お忙しいところお返事ありがとうございます。
調査する際のデータの提供など協力できるところがあればしますのでお気軽にご連絡ください。

またこちらで確認できたことですが、追加でもう少し細かくお伝えすると、「REAL ARCADE PRO V3-SA」の中の基板によってIONA-USで認識しないものも確認できました。
アケコンの背面を開けると基盤がありまして、

**Ver1.0では認識しました。
Ver1.2では認識しませんでした。**

[image]
[image]

どちらのアケコンも仲間内では所持しているものなのでもし何かあればご連絡ください。
重ねて、お返事ありがとうございました。

toyoshim, 2023-11-22:
情報ありがとうございます。
RAPはNが特殊で動作しない事がわかってはいたものの、Nだけ値段が高いので入手のふんぎりがつかず調査できていませんでした。SAでもバージョン違いで動かないのは新情報です。普段SA使ってるんですが、ほかにも仕舞い込んでるSAがあるので、そちらも含めて調べてみます。場合によってはログデータ収集のご協力をお願いするかもしれませんし、SA対応ついでにNも直れば良いのですが。
もしそちらのTwitterアカウントがわかれば、そちらでもDMできるようフォローします。

Sonoda-a, 2023-11-22:
ありがとうございます。
私のアカウントは**@akira_dayon**です。
ほぼ毎日System256を起動させていますので、ログデータ収集など何なりとお申し付けください。

よろしくお願いします。

toyoshim, 2023-11-22:
TwitterフォローしたのでDM通るようになったかと思います。
で、System256本体とIONAとの互換性ですが、最新版ファームウェアで確認したところ問題なく動作しました。
動作状況を整理すると、

Firmware Ver 2.17
IONA-US PCBv1、PCBv2、PCBv3にて確認
設定は初期値の１番をベースに、JVSデータ信号レベル補正をONにする必要があります（PCBv3では不要）

RAP側の問題については引き続き調べてみますが、System256の方は解決しそうなのでRAP用に別途バグを登録しました。
https://github.com/toyoshim/iona-us/issues/66

もしv2系ファームウェアに慣れていなかったら、以下の手順で試してみてください。
1. https://toyoshim.github.io/iona-us/firmware から2.17に更新する
2. このまま電源を入れなおすとLEDが3回ずつ点滅する状態になる。v2用ユーザーデータが見つからないエラー状態なので https://toyoshim.github.io/iona-us/setting から設定を書き込む必要がある
3. 設定ページの「デバイスを探す」からIONAを再度ファーム更新モードで接続し、データ初期化の確認に同意する
4. 設定１を対象として「JVSデータ信号レベル補正」をOFFからONに変更し「変更を確定」を押す
5. 「デバイスを探す」ボタンが「変更を保存する」ボタンに変わっているので、それを押してデータを保存する
6. もしボタン配置変更が必要な場合、v2ではIONA単体での設定ができないため、このページ内のボタン配置から変更し、「変更を確定」「変更を保存する」の手順を踏む（設定は6種類までIONAに保存でき、IONA単体ではこの6種類を切り替えることだけできます）

v3基板のうりの1つがnamco基板においても「JVSデータ信号レベル補正」設定を入れなくても認識するようになった点があげられます。複雑な設定をせずとも買ったままの初期状態でSystem256が認識するのではないかと思います。

Sonoda-a, 2023-11-23:
早速のご確認ありがとうございます。
以下で確認しました。

- Firmware Ver 2.17
- IONA-US PCBv2

JVSデータ信号レベル補正をONにしなければいけない認識がなくOFFのままでした。
ご連絡いただいた手順で実施したところ、System256で起動が確認できました。
しかし、コントローラが認識せず、IONAのTESTボタンを押下してもTESTモードに移動してくれなかったり、
ゲームもコントローラスタートボタンを押しても認識せず開始することができませんでした。
コントローラは「REAL ARCADE PRO V3-SA」のVer1.0です。

ログを採取してみたのですがご確認いただけますか？

[20231123_log.zip](https://github.com/toyoshim/iona-us/files/13445182/20231123_log.zip)

toyoshim, 2023-11-23:
確認ありがとうございます。
TESTボタンに関してはSystem2x6がDIPSW 1しか見ていないだけな気がしていますが、ちょっともう一度確認してみます。
RAPに関しては別途切り分けて調査しますので、今後は以下の方でご報告します。
https://github.com/toyoshim/iona-us/issues/66

こちらのレポートはTESTボタンの動作が1.23の頃と同じかどうかだけ確認してCloseにしたいと思います。
ほかに何か気づいた点があれば遠慮なくお願いします。

Sonoda-a, 2023-11-23:
ログでDIPSWのことも分かるのですね。
DIPSWは1のみ上げて残りは下げています。周りもその環境でやっています。

RAPは#66とのことで承知しました。失礼致しました。現状はファームウェア2.17でTESTボタンを押下してもテストモードに移動しない状態です。

よろしくお願いします。

Sonoda-a, 2023-11-23:
分かりました。
26日まで外出しているのでまた確認して連絡します！

Sonoda-a, 2023-11-26:
大変お待たせしました。
ご連絡いただいた項目でテストをしました。
結論から言うと3番目の項目で無事に動作しました。

1. ファームウェア1.23
	1. ファーム更新のページに1.23はなかったので1.24で試しました。
	2. System256起動時の「JM Master(2.25) Reset」の数字がくるくる回るだけでその先に進めませんでした。

2. 2.17 + JVSデータ信号レベル補正 ON
	1. 前回報告させていただいた結果と同じでした。
	2. https://github.com/toyoshim/iona-us/issues/65#issuecomment-1823687066

3. 2.17 + JVSデータ信号レベル補正 OFF
	1. 起動しました。TESTボタンも動作しました。

ファームウェア更新の項目に記載されている設定ページでの初期化がどれか分からず実施していなかったのですが、
もしかして「接続準備→設定」ということなのかな？と思いやってみたところ上手くいきました。
初期化はこれのことで合っていますか？

toyoshim, 2023-12-10:
1.23の件はすみません、古いファームウェアを整理した時に公開を止めていたようです。
ただ、確認の2と3から判断するに設定のON/OFFで対応できているようですね。

設定の初期化に関しては、LEDが3回点滅するのを繰り返す場合には初期化が必要で、「デバイスを探す」ボタンから接続した時に初期化して良いか確認するので、それを承認した後に、必要に応じて設定を変更し、最後に最初に押したボタンが「設定を保存する」ボタンに変化しているので、それを押して保存です。設定１～設定６のセットについては変更時には個別に「設定の確定」を押す必要があります。なので「デバイスを探す」「設定の確定」「設定を保存する」の３ステップになります。

という事で、問題なさそうなので、こちら側のバグはいったんcloseします。
もし追加で気になることがあればお知らせください。

closed 2023-12-10 by toyoshim