toyoshim/iona-us #71: Force Feedback Support for Naomi 2 Initial D

dannahan, 2024-03-08:
Not an expert on how this all works, but would like to see if Force Feedback is possible for Naomi 2. The data is coming from the MIDI out (CN9 on the Daughter Board).

Currently, there’s an open source program called OpenFFB (https://github.com/Fredobedo/openFFB)

However, you’ll need a "Sega FFB Controller" that you can get from Aganyte. (https://www.gamoover.net/Forums/index.php?topic=42310.0)
It works for PC and the WiPi

If there’s a way you can bypass the ffb controller hardware, and get the pinouts from the naomi MIDI into the second usb slot, would that work?

[image]

[image]

toyoshim, 2024-03-10:
thank you for the request and information. it's good to know that the naomi has a MIDI on the back panel and FFB uses it.
Now I spent some time to find an idea to support this, especially how we could wire.
So, my current idea is to use the IONA's P2 USB port for USB-MIDI.

NAOMI <=JVS=> IONA:JVS (as we do for all games)
NAOMI <=MIDI=> USB-MID conv <=USB=> IONA:P2 (extra wiring)

This may be a possible solution without modifying IONA-US, but needs an extra MIDI converter.

I could not work on this immediately, but let me have a middle term plan.

toyoshim, 2024-03-10:
An interesting tweet from Lord Hiro.
https://twitter.com/hiro_h10th/status/1090162225510150144

toyoshim, 2024-03-10:
Another possible wiring option would be

NAOMI <=MIDI=> Sega FFB Controller <=RS232=> USB-RS232 <=USB=> IONA:P2

I need to know Sega FFB Controller's output protocol, but this would help to support more titles via their conversion.

dannmanahan, 2024-09-16:
From Aganyte:
"here is the part of the code which manages the sending of the data serially. This should help you understand the data format."

[image]

English Translation
[image]

dannahan, 2024-09-16:
From Aganyte:
"here is the part of the code which manages the sending of the data serially. This should help you understand the data format."

[image]

English Translation:
[image]

dannahan, 2026-01-16:
Wanted to raise this up to see if there is anything needed to push this feature forward. Please let me know.