toyoshim/iona-us #78: Preset Contribution / プリセット募集

toyoshim, 2025-02-16:
(English)
If you are happy to share your preset for the IONA-US setting page, please post your preset (.jp) here.
The file can be obtained via the "Export to a File" button in the setting page.

Please add information about your expected controller and game titles.

I will peek this thread periodically, and try merging it to the setting page.

(日本語)
IONA-USの設定を共有しても良いよ、という人は、ここにプリセット（.jp）ファイルを投稿してください。
ファイルは設定ページの「ファイルに保存」ボタンから作成できます。

想定しているコントローラとゲームについての情報も添えてください。

定期的にチェックして設定ページに統合します。

toyoshim, 2025-02-16:
example - contribution from Kronos / 例 - Kronosさん提供

Use "Page, drop, or click to add files" to attach a file, but it should be compressed as zip as it cannot support uploading arbitrary file (i.e. our .jp file) directly. / 下の「Page, drop, or click to add files」からファイルを貼付してください。特定のファイルタイプしかアップロードできない（.jpファイルはアップロードできない）ようなのでzipで圧縮したものを貼付してください。

[Mario Kart Arcade GP2 _ DUALSHOCK 4.jp.zip](https://github.com/user-attachments/files/18814709/Mario.Kart.Arcade.GP2._.DUALSHOCK.4.jp.zip)

Controller: DUALSHOCK 4
Title: Mario Kart Arcade GP2

toyoshim, 2025-02-16:
I will put the launch icon when I merge it to the setting page like ↑.

pizzzza19, 2025-02-22:
基板：SEGA NAOMI
ゲームメーカー：SEGA
ゲーム名：THE HOUSE OF THE DEAD 2
コントローラー：Xbox Oneコントローラー
R2トリガーで撃つのでガンコンのフィーリングを再現しました。IONAの連射機能を使ってR1長押しでマシンガン（連射）も再現しています。実機には無いのであくまでチート扱い（オマケ）です。
基板の仕様でクロスヘアー（銃座標）は撃った瞬間のみ表示されます。アナログスティックを離すと座標は自動的に中心部に戻ります。
[MP7-IONA-US_HOD2.zip](https://github.com/user-attachments/files/18917993/MP7-IONA-US_HOD2.zip)

toyoshim, 2025-02-25:
thank you for your contributions.
I just imported 3 presets above.

Now I roughly expect that PS series controllers and Xbox series controllers can share the same setting.
So, let me say it's just "Gamepad" in the preset name. If we find a controller specific problem, I will update the name to be clearer.
Probably, DirectInput compatible controllers behave differently.

Also, I will update the setting page UI later so that I can show contributor's name and explanation for each entry.
That means your additional explanations like above are appreciated.

KronosMK7, 2025-02-26:
マリカアーケード2のプリセット提供したクロノスです。説明文も加える様でしたらこちらに置いておきます。

基板：Triforce
ゲームメーカー：NAMCO BANDAI Games Inc.
ゲーム名：マリオカートアーケードグランプリ2
コントローラー：DUALSHOCK 4

〜〜ボタン配列〜〜
○=アイテムボタン
×=対戦キャンセルボタン(右側にある赤いボタン)
SHARE=クレジット
Lスティック=ハンドル
L2=ブレーキ
R2=アクセル
十字キー↑↓=テストメニュー用カーソル※

※普段は使わない箇所ですが、テストメニューのみ動作し、ゲーム中に触れても害は無いみたいなのでおまけで設けてます

アナログ入力幅を16bitにしてる点に関して。自動にしてても大した差は有りませんが、NAOMI系統だとスティック操作が偶に不自然な挙動をする事がありますので、とよしまさんのご指示の元補正の意味を兼ねて変更しています

ーー使用する際の注意点ーー
トラブル防止の為、トライフォース側のテストメニューから以下の設定になっているか確認してください(※初代の場合は項目無いので不要)
SYSTEM MENU
↓
ENTER GAME TEST
↓
GAME OPTIONS
↓
ITEM BUTTON→BLUE&RED(初期設定)

アイテム使用する時通常はアイテムボタン又は対戦キャンセルボタンで発射しますが、例えばここをREDにしているとレース中対戦キャンセルボタンでしか発射出来ない現象が発生しますので原則上記で固定にしてください(詳しくはマリオカートアーケードグランプリ2 取扱説明書49P参照)

toyoshim, 2025-03-01:
ここまでのプリセットと説明を反映しました。
説明は、ひとまずメニューのアイテムの上でホバーしているとTooltipとして表示されるようにしてみました。
もっとこうした方が嬉しいとか希望があれば教えてください。
あと、クレジットは昔に提供してもらったものが「とよしま」表記になっちゃってるのがあると思うので、適宜「これは私が提供したもの」とクレームして頂けたらありがたいです。間違いの訂正とかもここで報告お願いします。

Now all presets and descriptions are merged.
As an initial implementation, descriptions will be shown as a tooltip when your mouse hovers over each menu item.
If you have an idea or preference, please let me know.
There would be mistakes on credits or descriptions. If you find some problems, or wrong credit on your contribution, please let me know. You can also report errors, or propose a revision here.

pizzzza19, 2025-05-26:
>間違いの訂正とかもここで報告お願いします。
プリセットの盛り込み・反映ありがとうございました。軽微なtypoと思いますがプリセット名が「The House Of The **Head** 2」となっていたので次回更新時に修正いただければ幸いです。別にこのままでも読み取れると思いますが一応内容見ましたということで報告いたします。

P.S.
合わせて一部内容の補足ですが銃のクロスヘアーは一般的な「+」ではなく弾痕のような「☆」マークとなります。どちらにせよ常時表示はされないのでピクセル単位で部位を狙っていくハイスコアプレイは実基板同様に困難です。（備忘で追記しました）

jpadillarando-gif, 2026-10-24:
Buenas me llamo Juan tengo una naomi con el iona jvs to usb y da error io sega marine fishing he probado varios presets pero siempre da error la cabina esta en standar y nada. 
Por favor alguna solución muchas gracias

nipple77, 2026-10-25:
Change the jvs type to a type 2 jvs board. 
Marine fishing needs the type 2 to work 

Cheers