toyoshim/iona-us #83: Allows to map service/test as any button

Rolel, 2025-03-23:
Hello,

From the setting page it's not possible to defince a gamepad button as "test" or "service". This would be nice to have.

Regards

toyoshim, 2025-03-24:
Technically this would not be so difficult, but unfortunately, the persistent memory in the device is just 1KB, and currently the whole area is already used for the 6 configurations (10B for common + 196B for each set x6 = 1024B)
Currently, each button can be assigned to 16 buttons, and this is represented as a 16-bit bitmap pattern. So, if I'd add one or two more bits, I need to expand this to be 32bit bitmap. So, that will reduce possible configuration set to be halved. Another concern is processing speed. It will double the processing time. So, unless this is critical use case, I'm inclined not to implement this for now.

toyoshim, 2025-03-25:
Yes, these are possible options and they all have a downside. So, I cannot put a priority on this request.
Can you tell me your use cases? I'm not sure why this is needed, and if this is the best solution for your requirement.

Rolel, 2025-03-25:
First use case that comes to my mind is easier navigation from the pad. When configuring boards in the service menu it's sometimes a mix of up/down button + service/test. It's also a pain to keep "test" pushed while holding the pad in another hand to navigate.

Last but not least, this would distinct select+test from the board itself so we can click both without changing settings.

toyoshim, 2025-03-25:
I just remember that the test button is not exposed to the JVS I/O.
So, technically it's impossible. The system board just checks their embedded TEST button and doesn't ask it to the I/O board.
Service button is defined, but actually I reuse the bit for the coin signal.

toyoshim, 2025-03-25:
Oops, sorry, the previous comment was wrong. Please ignore it. 
Yep, the TEST button is mapped in a different cycle.
So, possible idea to implement this is to omit button 9 and 10, and reuse these areas to the TEST and SERVICE.